leandrovip/Vip.Printer #10: ESC-POS.NET

leandrovip, 2019-03-19:
Hey Leandro,

Did you try out my library, ESCPOS_NET?  I'd be interested to hear your thoughts and see if you would be interested in using / contributing to my library.  It's pretty straightforward to use.

_Originally posted by @lukevp in https://github.com/leandrovip/Vip.Printer/issues/2#issuecomment-474177884_

leandrovip, 2019-03-19:
Hey @lukevp, how are you?
I do not test your library, but i will try this as soon as possible and give you feedback, ok? I began this project because was necessary to use esc/bema commands.

lukevp, 2019-03-19:
Great! What is esc/bema and how's it different from esc/pos?

leandrovip, 2019-03-19:
ESC/BEMA is a command schema for [Bematech](https://www.bematech.com.br/produtos/equipamentos/) printers. It´s similar in some commands (binary), but have also peculiarities.

lukevp, 2019-03-19:
Interesting!  This could be implemented in my library by creating another ICommandEmitter that can produce ESC/BEMA instead.  I don't have one of these to test, unfortunately.

leandrovip, 2019-03-19:
Do you use RawPrinterHelper?

closed 2019-03-20 by leandrovip