leandrovip/Vip.Printer #13: Utilização em WebForms

DanPrado77, 2019-06-09:
É possível usar essa lib em webforms? Tenho a necessidade de imprimir etiquetas de códigos de barras (code39) via web.

Abraços

leandrovip, 2019-06-09:
Não testei mas creio que seja possível sim. Qual impressora você utiliza? A lib usa RawPrint pra fazer as impressões então basta ela estar instalada ou passar o caminho que imprime.

[]’s

DanPrado77, 2019-06-10:
Bom dia!
Usamos a Elgin, temos uma aplicação web como front. Mas testei a lib e
tomei pau na locaweb. Erro de partial trust. Já tomamos esse erro com RDLC.

[]s

leandrovip, 2019-06-13:
Olá! Que pena.
Acredito que possa ser pelo fato de usar RawPrint.
Mas, de qualquer forma, caso eu possa ajudar em algo é só avisar aqui!

[]´s

[]´s

closed 2019-06-17 by leandrovip