leandrovip/Vip.Printer #14: WebForms

DanPrado77, 2019-06-18:
Fala Leandro,
Tudo bem? Tentei aplicar esse mesmo componente em uma aplicação web. E não rolou!

Ele tenta mapear a impressora do server. Alguma sugestão?

leandrovip, 2019-06-18:
Opa! Tudo certo e contigo?

Qual o seu cenário? A impressora está instalado no client ou no server? 
A lib trabalha com o RawPrint, então faz comunicação direta com o spooler de impressão do **windows**.

No parâmetro nome da impressora, o que está informando?

Abs.

DanPrado77, 2019-06-18:
Opa!
Então, a impressora fica no cliente. E pelo que notei, o erro ocorre por
que ele busca a impressora no servidor.

[]s

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Livre
de vírus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>.
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

leandrovip, 2019-06-19:
Sim, creio que este é o comportamento sim.

Só para eu poder entender, qual é sua estrutura? Qual parametro está passando no nome da impressora?

DanPrado77, 2019-06-19:
Opa!
Se rodo local, passo o nome da impressora instalada e ele envia o documento
pra ela normalmente. Mas no servidor, não tenho como passar o nome da
impressora, pois a mesma fica na rede local e não no server.

[]s

brasvidal, 2019-06-19:
Não entendo muito, posso estar enganado, mas trabalhando com web, não seria melhor gerar um pdf e mandar para impressão? A não ser que tenha um cliente escutando as requisições da impressao, não vejo como usar isso.

Marcos Vidal
Analista de Sistemas
Casters Automação Comercial
(45)3523-6012
(45)9903-6551

leandrovip, 2019-06-19:
Jóia! Então nesse caso, compartilhe a impressora e passe o nome do compartilhamento que pode dar certo
Por exemplo:

```
\\NomeDoPc\NomeDoCompatilhamento
```

[]´s

leandrovip, 2019-06-22:
@DanPrado77, jóia?

Quando puder me dê um feedback se com o nome do compartilhamento deu certo! 
Fiz um teste aqui e deu certo! 

[]´s

leandrovip, 2019-06-27:
Em caso de dúvidas, por favor, fique à vontade para reabrir esta issue ou abrir uma nova!

[]´s

closed 2019-06-27 by leandrovip

DanPrado77, 2019-06-27:
Opa! Tudo bem Leandro!
Então, não deu certo. Nossa aplicação roda num servidor de hospedagem web
(Kinghost). E acessamos via browser. Não conseguimos compartilhar a
impressora para fora de nossa rede. Então mesmo compartilhando, não é
possível chamá-la do servidor.

:-(

[]s