leandrovip/Vip.Printer #15: Hoje em dia é possível imprimir diretamente do browser?

EwertonDutra, 2019-08-02:
Acompanhando um pouco as API dos browser, tenho visto que cada vez mais a web está ganhando permissões que antes não era permitido.
No site: https://permission.site mostra que é possível ter acesso ao USB

Seria então possível um site ter acesso a uma impressora não-fiscal (exemplo a 4200) ?

leandrovip, 2019-08-09:
Muito provavelmente sim.
Mas não tenho esta informação com certeza.

Precisaria de testar. Caso teste, por favor, informe aqui pra nós.

[]´s

EwertonDutra, 2019-08-11:
A informação mais relevante sobre o assunto até então que achei foi:
https://www.qz.io/
O link acima mostra um jeito de fazer o que perguntei.
Ainda não funciona de forma tão prática, do estilo **"só fazer o site e funciona em tudo"**, mas com algumas instalações e tudo vai funcionar :) 

Espero que ajude outras pessoas também.

leandrovip, 2019-08-12:
Show, Ewerton!
Obrigado por compartilhar!

[]´s

closed 2019-08-21 by leandrovip