leandrovip/Vip.Printer #19: Implementação para Sat/NFCe

marcosgerene, 2020-05-07:
@leandrovip 

Vi na sua wiki que você já atende uma gama interessante de impressoras o que facilita muito as coisas.

Tenho intenção de implementar a impressão da NFCe e Sat em esc/pos.

Se eu realmente iniciar o processo, pretendo usar como "base" os seguintes projetos:
 - https://github.com/ZeusAutomacao/DFe.NET para NFCe
- https://github.com/ACBrNet/ACBr.Net.Sat para Sat

Ambos são MIT (assim como o seu), logo acredito que até a questão da licença vai ficar legal.

Bem, mas antes de começar, me diz se você já tem algo encaminhado para não ter que "reinventar a roda" a toa.

Att, Marcos

marcosgerene, 2020-05-07:
https://github.com/marcosgerene/Gerene.DFe.EscPos

closed 2020-05-07 by marcosgerene

leandrovip, 2020-05-09:
@marcosgerene 

Desculpe a demora.
Tinha o layout do SAT, mas vi em seu projeto que está basicamente igual ao o que tenho aqui, rs. 

Bom, qualquer coisa só avisar!
[]´s

Att,