leandrovip/Vip.Printer #20: Implementar impressão do Logotipo

leandrovip, 2020-05-08:
Disponibilizar método para impressão do logotipo para EscPos e EscBema.

marcosgerene, 2020-07-10:
@leandrovip 

Existe alguma novidade/previsão sobre isso ?

leandrovip, 2020-07-10:
@marcosgerene,

Já tenho uma implementação sobre isso, porém estava sem impressora para testes. Chegou uma hoje. Vou fazer alguns testes e disponibilizo.

marcosgerene, 2020-07-10:
@leandrovip 
Tenho uma i9 e uma tm-20 para testar.

O @adrbarros tem DR700 lá

adrbarros, 2020-07-11:
Aqui temos
Elgin i9, Elgin i7 (Escritório)
Epson TM-T20 (Escritório)
Daruma DR700 (Escritório)
PrintID (Escritório)
Tanca TP-650 (Cliente)
Bematech MP 2500, 4000, 4200 (Cliente)
Agora adiquirimos uma pra testes bem baratinha marca EVADIN (Padrão ESC/POS)
Se precisar fazer testes, as impressoras que não temos no escritório pegamos nos clientes

closed 2020-07-16 by leandrovip