leandrovip/Vip.Printer #25: Multi Language Support

San-Nabaty, 2020-05-15:
Can this print non english/Latin text like Arabic and Hebrew ?

leandrovip, 2020-05-16:
Hi,
I think so. What is the encoding your printer to use?

Regards;

San-Nabaty, 2020-05-16:
I don't have much info about it but what i know it can support non latin if it's sent properly , for example Arabic uses Unicode encoding.

leandrovip, 2020-05-16:
Yes.
If your printer has encoding Unicode set, it's possible to set the encoding in Printer.cs at line 154.
Can you make a test?

Regards;

San-Nabaty, 2020-05-16:
check your inbox leandro@vipsolucoes.com I sent you more details

leandrovip, 2020-05-16:
Hi, 

I added a new method Append with Encoding property, can you a test at branch **teste-encode**? 👍 

```c#
            var printer = new Printer("PrinterName", PrinterType.Epson);

            printer.Append("Test UNICODE", Encoding.Unicode);
            printer.Append("Test ASCII", Encoding.ASCII);
            printer.Append("Test ASCII", Encoding.UTF8);
            printer.NewLines(3);
            printer.Append("End test :)");
            printer.PartialPaperCut();

            printer.PrintDocument();
```

Regards;
Leandro

San-Nabaty, 2020-05-16:
Hi, 

I think I will need to have the updated version in order to make the test right ? or did you upload the updated version?  also where is exactly teste-encode ? 

~San

leandrovip, 2020-05-16:
Yes, 

https://github.com/leandrovip/Vip.Printer/tree/test-encode

leandrovip, 2020-05-28:
Anything just let me know.

Regards,
Leandro

closed 2020-05-28 by leandrovip