leandrovip/Vip.Printer #29: Uma dúvida sobre o uso da lib

EwertonDutra, 2020-07-16:
Boa noite

Olha eu de novo ai @leandrovip , recentemente integrei uma impressão com a Epson TM-T20 e depois precisei de integrar com a Daruma DR800

Lembrei desta lib
Gostaria de tirar duas dúvidas sobre o uso
1° para usar é só deixar junto ao meu executável as DLLs de cada impressora? Se não quais seriam os requisitos?
2° Já tenho a variável com o ESC/POS pronto como usaria nesta lib?

`using Vip.Printer;
            var printer = new Printer("MP4200", PrinterType.Epson);   // >>> qual seria o do ESCPOS?

            printer.Append(varWithEscPos);
            printer.PartialPaperCut();
            printer.PrintDocument();
`

A DR800 não foi testada ainda não? eu teria uma forma de testar em um cliente e posso dá o feedback

Obrigado desde já

leandrovip, 2020-07-16:
@EwertonDutra, tudo certo?

1) Sim, para distribuir é apenas deixar a dll Vip.Printer junto ao executável. Faz a instalação da biblioteca via NuGet e ao compilar seu projeto ela já estará na pasta do executável. A biblioteca não utiliza uma dll para cada tipo de impressora, ela é baseada nos sets de comando das impressoras (esc pos, esc bema e esc daruma). Então isso é abstraído de forma que você possa utilizar o mesmo código, ou seja, a mesma o mesmo layout para impressoras diferentes.

2) O que seria a váriavel com o EscPos? Como disse na resposta anterior, depois de feito o layout o que será alterado é apenas a forma com que o objeto Printer é instanciado, com o nome da impressora/porta/compartilhamento e o set de comando desejado. Deu uma olhada no projeto de demonstração?

3) A DR800 em específico creio que não, apenas a DR700 (que eu tenha conhecimento). Mas provavelmente terá as funções principais funcionando.

Qualquer outra dúvida, só avisar.

[]´s

EwertonDutra, 2020-07-16:
Tudo certo sim. Espero que com você também.

**1.** Entendi, legal!

**2.** 
```
string szTexto = new String(' ', 4096);
szTexto = "<ce><da><b>" + pedido.usuario.nome + " | " + pedido.usuario.telefone + "</b></da></ce>\n";
```

-> Seria uma variável com comandos ESC/POS   ;)  igual acima
Sobre o projeto olhei navegando pelo git mesmo, e criei um projeto rápido instalando a lib pelo NuGet e olhei os autocompletes depois que adicionei pelo using a lib.

**3.** Obrigado posso testar e dá o feedback aqui

leandrovip, 2020-07-16:
Jóia!

Sim, no caso a formatação seria diferente. Depois de instanciar o objeto ele te dá alguns métodos que você pode formatar da forma que deseja, por exemplo.

```csharp
var printer = new Printer("NomeImpressora", PrinterType.Epson);

printer.AlignCenter();
printer.WriteLine("Meu texto centralizado aqui")

printer.AlignLeft();
printer.WriteLine("Meu texto alinhado a esquerda");

....
printer.PrintDocument();
```

Desta forma você iria montando seu layout.
Certo, qualquer dúvida só postar aqui!

[]´s

EwertonDutra, 2020-07-16:
Entendi

no caso depois do 
`printer.WriteLine("Meu texto alinhado a esquerda");`
A formatação é resetada ?

Digo porque se eu colocar centralizado e negrito (em uma linha no caso)
E mudo o alinhamento (tipo left) na próxima linha já altera ... **mais o negrito tenho que colocar ele normal?** ou zera depois do **WriteLine** ?

leandrovip, 2020-07-16:
No caso do alinhamento você precisa voltar, ou seja, seta centralizado > imprime > seta esquerda/direita.
Para funções de fonte como negrito, itálico, etc você tem duas opções.

1) Usa o método bold e passa o texto que será apenas negrito

```charp
printer.BoldMode("Texto com negrito");
printer.WriteLine("Texto sem negrito");
```
2) Usa o método bold para habilitar o negrito > imprime > desabilita

```charp
printer.BoldMode(PrinterModeState.On);
printer.WriteLine("Texto em negrito");
printer.WriteLine("Texto também em negrito");
printer.BoldMode(PrinterModeState.Off);

printer.WriteLine("Texto normal");
```

EwertonDutra, 2020-07-16:
Entendi show
Bem pensado
Vou usar no projeto aqui

No caso estes comandos novos não estão no NuGet ainda não né? 
Tuesday, May 12, 2020 (5/12/2020) última versão lá

leandrovip, 2020-07-16:
Não, está no PR.. vou atualizar hoje.

EwertonDutra, 2020-07-16:
Entendi, ok

@leandrovip  Obrigado ai por tirar as dúvidas

closed 2020-07-16 by EwertonDutra

leandrovip, 2020-07-16:
Já está disponível! 
Valeu, abs.

EwertonDutra, 2020-07-16:
Já atualizei rrsrs