leandrovip/Vip.Printer #35: [Code128] Problema com impressão de barras

marcosgerene, 2020-09-15:
No projeto https://github.com/marcosgerene/Gerene.DFe.EscPos que usa este como base "motor" para impressão de DFes, encontrei um problema na leitura do código de barras do SAT. 

Segue xml. Você pode usar qualquer demo para testar.

Att, Marcos
[CFe35200911111111111111591234567890001071072281.zip](https://github.com/leandrovip/Vip.Printer/files/5224705/CFe35200911111111111111591234567890001071072281.zip)

marcosgerene, 2020-09-26:
@leandrovip 
Alguma novidade sobre o bug? Posso ajudar de alguma forma? (Tenho uma TM-20 e uma Elgin i9 para testes).

leandrovip, 2020-09-27:
> @leandrovip
> Alguma novidade sobre o bug? Posso ajudar de alguma forma? (Tenho uma TM-20 e uma Elgin i9 para testes).

@marcosgerene, 
Por enquanto não tenho novidade, fiquei ausente alguns dias (minha filha nasceu!) e não consegui ver nada sobre este problema. Volto esta semana e já verifico o que está acontecendo.

marcosgerene, 2020-09-27:
@leandrovip 

Parabéns cara! Desejo muita saúde para sua filha, abração

leandrovip, 2020-10-16:
@marcosgerene 

Poderia fazer um teste (#36) antes de eu subir a atualização?
Fiz o teste em uma ElginI9 e Dimep Dual, nas duas imprimiu e leu corretamente, caso ok já faço o merge.

Obrigado

marcosgerene, 2020-10-16:
@leandrovip 

Cara, não vi sua resposta, me desculpa... eu só consigo na segunda-feira, estou sem impressora aqui..

leandrovip, 2020-10-17:
Tranquilo, aguardo!

[]´s

leandrovip, 2020-10-20:
Fiz o teste em outras impressoras e o código foi impresso corretamente, mas, qualquer coisa só avisar!

[]´s

closed 2020-10-20 by leandrovip

marcosgerene, 2020-10-21:
@leandrovip 

Fiquei fora ontem e hoje, implantação.

Vou testar e te dou o feedbac, abraço