leandrovip/Vip.Printer #37: Impressão Ethernet

FranchiAle, 2020-10-21:
Bom dia
Como proceder para impressão pela porta ethernet?
Tenho a TP-650 configurada para impressão ethernet e não encontrei exemplos de como proceder para imprimir diretamente para o IP da impressora.
Tentei destas formas:
var printer = new Printer(@"\\192.168.1.180", PrinterType.Epson);
var printer = new Printer(192.168.15.180", PrinterType.Epson);
Mas não obtive sucesso. É possível imprimir diretamente para o IP?!

leovictorcvo, 2020-10-21:
Oi, tudo bem?
Então, estou passando o mesmo problema com uma Daruma DR700 ETH. Consegui contornar utilizando o Driver Spooler que a Daruma fornece. Mas se conseguisse imprimir diretamente pelo ip da impressora seria melhor.

leandrovip, 2020-10-21:
@leovictorcvo, @FranchiAle 

Atualmente a biblioteca só conta com a impressão via spooler, sendo assim, por enquanto, faço o mesmo 'contorno' que o @leovictorcvo faz, instalo o driver genérico mesmo da microsoft apontando para o IP da impressora. Não é o ideal, mas por hora supre esta necessidade.

Porém, vou analisar a possibilidade de implementar uma outra interface para impressão direta na porta.

[]´s

FranchiAle, 2020-10-21:
@leandrovip , agradeço pela ajuda!

rftd, 2020-11-13:
Cara a impressão por ETH é basicamente uma conexão TCP, precisa do IP e da porta apenas.

thiagovalim, 2021-02-05:
Este projeto tem um exemplo

https://github.com/lukevp/ESC-POS-.NET/blob/master/ESCPOS_NET/Printers/NetworkPrinter.cs

closed 2021-07-03 by leandrovip

leandrovip, 2022-04-18:
#56