leandrovip/Vip.Printer #39: Erro no método WriteLine

thiagovalim, 2021-02-05:
Se você usar o método NewLine() ele pula duas linhas e não apenas uma, devido ao fato que o método NewLine chama o método WriteLine que também adiciona o caracter \n

public void NewLine()
        {
            WriteLine("\n");
        }

leandrovip, 2021-02-05:
@thiagovalim,

Observe a imagem em anexo:
Na impressão 1 está como atual, com o `WriteLine("\n")`, tenho o retorno como o esperado, é pulado 1 linha 'completa', e não somente alterado o 'carro de impressão' para a linha de baixo.

Na impressão 2, alterei o método NewLine() para `WriteLine("")` e já não tenho o retorno esperado, a impressão é feita na linha debaixo, e não ocorre o pulo completo como é o esperado por este método.

Obs: Testes em uma Dual Print DIMEP, Bematech 2500, Elgin I9.

Qual o retorno que gostaria obter? Em qual modelo de impressão você fez o teste?

[image]

[]s

thiagovalim, 2021-02-05:
No caso eu fiz o teste na bematech MP 4200 e daruma DR800 o método NewLine() pula duas linhas seguidas semelhante a primeira imagem (pulou duas linhas entre a linha 2 e 3, pois ao chamar o método NewLine() ele chama o método WriteLine() com \n ficando: \n\n o que faz com que pule duas linhas

leandrovip, 2021-02-05:
Jóia, vou fazer um update.

leandrovip, 2021-02-05:
Atualizado, obrigado!
[]´s

closed 2021-02-05 by leandrovip