leandrovip/Vip.Printer #40: Impressão de Logo

thiagovalim, 2021-02-05:
Olá existe alguma função para impressão do logo pré carregado na memória da impressora?  Eu também percebi que a impressão do logo através do método Printer.Image funciona perfeitamente com bematech e epson mas não funciona com impressoras daruma.

leandrovip, 2021-02-05:
Olá, @thiagovalim 

Não tem a função para impressão da logo na memória, a biblioteca não faz este gerenciamento, pelo que eu vi, em algumas impressoras isso é bem particular, então resolvi implementar somente este modelo de impressão que atende a maioria dos modelos, pelo menos por enquanto.

Sim, pode ter alguma diferença em impressoras daruma sim, pois quando implementei não consegui nenhuma impressora daruma para fazer os testes necessários. Se tiver disponibilidade de testar nestes modelos, será de grande ajuda.

[]´s

thiagovalim, 2021-02-05:
Olá @leandrovip, eu verifiquei esta questão já e é bem simples e funcionou perfeitamente, tem dois comandos para impressão do logo armazenado na memória. O mais antigo compatível a maioria dos modelos que usam ESCPOS e o mais novo usado nas novas impressoras da epson (este não tem tanta compatibilidade) o mais recomendável é utilizar o mais antigo. Segue abaixo:

Comando Antigo:
FS + 'p' + n + m 
n = indice (0 - 9)
m = opcoes de impressao (1 - 3) 1 é normal, 2 expandido

Comando novo:
GS + '(L' + #6 + #0 + #48 + #69 +KeyCode1 + KeyCode2 + FatorX  + FatorY
este tem mais opções, mas é menos compatível

o primeiro foi testado com sucesso através da função Write (testei na bematech e daruma, vou testar na tanca hoje)
a impressão ficou muito mais rápida e nítida do que carregar a imagem dinamicamente

thiagovalim, 2021-02-05:
@leandrovip  Quanto a impressão da imagem na daruma, não está funcionando mesmo, sai apenas um pedaço da imagem e buga, fazendo a impressora imprimir várias linhas em branco sem parar, depois vou ver se consigo ajustar, se eu fizer os ajustes com sucesso eu posto

leandrovip, 2021-02-05:
> Olá @leandrovip, eu verifiquei esta questão já e é bem simples e funcionou perfeitamente, tem dois comandos para impressão do logo armazenado na memória. O mais antigo compatível a maioria dos [...]

@thiagovalim,
Sim, a impressão realmente deve ficar mais rápida pelo fato de já estar na memória. 
E para setar a imagem na memória da impressora? Somente através dos softwares do fabricante?
Mas sim, é uma opção, assim que eu conseguir vejo de implementar um método para esta forma de impressão.

thiagovalim, 2021-02-05:
@leandrovip  dá pra setar por comandos também, assim com a impressão existem dois comandos o antigo e o novo com mais opções

thiagovalim, 2021-02-05:
Comando antigo para definir imagens FS q (Define NV bit image), tem o novo com mais opções só que é menos compatível

marcosgerene, 2021-06-05:
@thiagovalim 

Se não estou enganado o @adrbarros implementou as Darumas.

Ele tem algumas para testes, acredito que deva ter testado exaustivamente a impressão de logotipo... marquei ele aqui porque pode ajudar em algo, mas não seria interessante checar também algum tipo de atualização para a impressora?

adrbarros, 2021-06-05:
Olá a todos @marcosgerene, @leandrovip, @thiagovalim. Acabei de testar aqui impressões na Daruma Dr700 Le e Dr800 aparentemente OK. Não implementei a impressão de Logo gravado na memória da impressora. Vou tentar dar uma olhada

closed 2021-07-03 by leandrovip