leandrovip/Vip.Printer #44: Dúvidas

meciasbueno, 2021-08-02:
Preciso fazer impressão de recibo em impressoras térmicas não fiscais, tenho algumas dúvidas com relação a essa lib:

- Preciso enviar alguma dll específica do fabricante da impressora junto com a minha aplicação?
- Eu não tenho a impressora para testar, podem me recomentar um emulador?

leandrovip, 2021-08-02:
@meciasbueno,

1) Não, não é preciso enviar qualquer arquivo do fabricante junto com sua aplicação, apenas a lib mesmo. Como foi desenvolvida em cima dos comandos de impressão.. não faz dependência de arquivos do fabricante.

2) Até hoje não encontrei um emulador efetivo para este cenário, sempre testei com a impressora mesmo. Como você não tem, é possível você conseguir uma por empréstimo com os fabricantes. É possível com a Gertec, Elgin, Tanca, Control ID, Dimep.. todas tem um programa de apoio ao desenvolvedor. Vale a pena, pois testando com a impressora do lado é muito mais efetivo o desenvolvimento. Se bem que com a lib o desenvolvimento é bem produtivo.

[]´s

meciasbueno, 2021-08-03:
> @meciasbueno,
> 
> 1. Não, não é preciso enviar qualquer arquivo do fabricante junto com sua aplicação, apenas a lib mesmo. Como foi desenvolvida em cima dos comandos de impressão.. não faz dependência de arquivos do fabricante.
> 
> _Edit: **Ressalva**, a não ser que queira enviar os arquivos de instalação da impressora. O que dependendo do cenário também não é preciso. A maioria das impressoras de hoje já se auto reconhecem (drivers), o que faço é adicionar a impressora como "Genérica/Somente Texto. Mas isso vai depender do seu ambiente._
> 
> 1. Até hoje não encontrei um emulador efetivo para este cenário, sempre testei com a impressora mesmo. Como você não tem, é possível você conseguir uma por empréstimo com os fabricantes. É possível com a Gertec, Elgin, Tanca, Control ID, Dimep.. todas tem um programa de apoio ao desenvolvedor. Vale a pena, pois testando com a impressora do lado é muito mais efetivo o desenvolvimento. Se bem que com a lib o desenvolvimento é bem produtivo.
> 
> []´s

Entendido, muito obrigado pelos esclarecimentos.

closed 2021-08-04 by leandrovip