leandrovip/Vip.Printer #45: Impressora para teste

meciasbueno, 2021-08-04:
Pode me dar uma dica? A impressora abaixo é uma boa opção para eu comprar para usar no desenvolvimento?

[https://produto.mercadolivre.com.br/MLB-1949658995-mini-impressora-portatil-bluetooth-termica-58mm-android-ios-_JM?matt_tool=70509262&matt_word=&matt_source=google&matt_campaign_id=12408848785&matt_ad_group_id=119017168518&matt_match_type=&matt_network=g&matt_device=c&matt_creative=500385009285&matt_keyword=&matt_ad_position=&matt_ad_type=pla&matt_merchant_id=414912031&matt_product_id=MLB1949658995&matt_product_partition_id=296321646450&matt_target_id=aud-879604627128:pla-296321646450&gclid=CjwKCAjw9aiIBhA1EiwAJ_GTSq3VJ0vloSOAhAAaMuq49RrNFVPx16rFlaRyiRE5xyjhHqaZ_LXM2hoCLY8QAvD_BwE](url)

leandrovip, 2021-08-04:
Para testes funciona bem, mas, qual intensão do comprovante? 
Digo porque para comprovantes tipo NFCe, CFe.. acho que fica melhor uma com 80mm.

Mas para testes, para desenvolvimento.. funciona bem sim.

meciasbueno, 2021-08-05:
Seria para imprimir comandas para um sistema de delivery.
Acredito que os usuários vão comprar a impressora tamanho normal, com 80mm.
Como eu estou começando agora com esse sistema, eu não queria gastar muito com isso.
Seria mais pra eu me basear mesmo, afinal a impressão vai ocupar todo espaço proporcional ao tamanho que o usuário estiver usando né?

leandrovip, 2021-08-05:
> ... afinal a impressão vai ocupar todo espaço proporcional ao tamanho que o usuário estiver usando né?

Não necessariamente. Na biblioteca existe algumas propriedades que guarda o número de posições limite para impressão, caso não informado nenhum valor na instancia é assumido um valor padrão. Você pode se basear neste valor para limitar a quantidade de caracteres impressos.
Caso você envie para impressora uma quantidade de caracteres que exceda o limite disponível da impressora, o texto que ultrapassar será impresso na linha de baixo. Então a biblioteca lhe da o controle de criar qualquer layout que quiser.

meciasbueno, 2021-08-05:
> será assumido um valor padrão

Os valores padrão são baseados em uma impressora 80mm?

leandrovip, 2021-08-05:
Sim, você pode alterá-los ao instanciar a classe Printer.

meciasbueno, 2021-08-05:
> Sim, você pode alterá-los ao instanciar a classe Printer.

Perfeito, muito obrigado Leandro. Vou fazer a compra dessa impressora.

closed 2021-08-05 by leandrovip