leandrovip/Vip.Printer #47: Alinhamento left/right na mesma linha

meciasbueno, 2021-08-10:
Como eu faria o alinhamento para a segunda parte do texto ficar alinhado á direita?
Ex: O "Total"  alinhado a esquerda e o "100,00" alinhado á direta.

```
  _printer.AlignLeft();
  _printer.WriteLine("Total: ");

  _printer.AlignRight();
  _printer.WriteLine("100,00");

```

leandrovip, 2021-08-10:
Geralmente eu faço isso já na string e mando somente uma linha sem precisar de alinhar via impressora, mas você pode tentar o seguinte:

```csharp
  _printer.AlignLeft();
 \\ O comando Write mando o texto para a impressora, mas não manda o comando de voltar a linha no final. Já o WriteLine faz este papel.
  _printer.Write("Total: "); 

  _printer.AlignRight();
  _printer.WriteLine("100,00");

```

meciasbueno, 2021-08-11:
> ```cs
>   _printer.AlignLeft();
>  \\ O comando Write mando o texto para a impressora, mas não manda o comando de voltar a linha no final. Já o WriteLine faz este papel.
>   _printer.Write("Total: "); 
> 
>   _printer.AlignRight();
>   _printer.WriteLine("100,00");
> ```

Dessa forma fica tudo alinhado á direita.
Bom, vou procurar uma forma de resolver via texto mesmo.

leandrovip, 2021-08-12:
Tenho um outro pacote com métodos de extensão que utilizo. Nele tenho um método que faz isso que você precisa, juntar duas strings com possibilidade de ficar uma ao extremo da outra. Você pode fazer um teste se quiser. Serial algo mais ou menos assim:

Instala o pacote via nuget
`install-package Vip.Extensions`

O código ficaria algo como:

```csharp

var printer = new Printer("MP2500", PrinterType.Bematech); 

var texto1 = "Meu texto 1";
var texto2 = "Meu texto à direita";
var textoParaImpressao = texto1.JoinStringAndFill(texto2, ' ', printer.ColsNomal);

// A variavel printer.ColsNormal contém um valor padrão, caso não alterado, de colunas permitidas para impressão. Então o método irá juntar as duas strings e preencher o espaço até o limite com um 'espaço' em branco neste cenário. 

printer.WriteLine(textoParaImpressao);
printer.PrintDocument();

```

closed 2021-09-01 by leandrovip