leandrovip/Vip.Printer #49: A elgin i9 usa o type = PrinterType.Epson

EwertonDutra, 2021-08-25:
Bom dia,
Olá @leandrovip 

Me confirma uma informação por favor, **a Elgin usa o type: PrinterType.Epson**   certo?
Vi algunas exemplos e está com esse type
Só por curiosidade (ou alguma coisa que não me atentei ainda) _porque a Elgin não tem um type próprio_? Elas internamente usam o mesmo comandos no caso?

Obrigado desde já

leandrovip, 2021-08-25:
Olá! 
Sim, PrinterType.Epson = EscPos, tem este nome porque a epson que criou este padrão e é o que vem sendo utilizado pela maioria das impressoras. Para Bematech e Daruma, em alguns modelos, é possível setar em suas configurações para utilizar o set EscPos, porém como existem algumas particularidades do próprio modelo, resolvi criar o estes set´s de comando separado. 

Creio que a tendencia, para novos modelos de impressora/marca é continuar utilizando o EscPos como set de comando principal, portanto o PrinterType.Epson seria utilizado ✌

[]´s

EwertonDutra, 2021-08-25:
Valeu 👍

closed 2021-08-25 by EwertonDutra