leandrovip/Vip.Printer #50: [Code128] Problema com impressão

ficodigital, 2021-11-30:
Olá, tudo bem!?

Ao efetuar um teste com o projeto Vip.Printer.Demo, para a impressão de código de barras, não funciona a impressão do código de barras método "Vip.Printer.Code128()", estou precisando imprimir o código de uma chave NFCe, mas sem sucesso... 

Notei que quando passo 20 caracteres o método funciona, porém ao passar 22 caracteres, passa a não funcionar...

[image]

Dufayel, 2021-11-30:
Boa tarde!

Eu tive o mesmo tipo de problema em alguns modelos de impressoras, para esse tipo de caso é necessário mudar o parâmetro de largura das barras, para isso acabei customizando o código da seguinte forma na classe **BarCode**.

`var barWidth = (byte) (code.Length <= 22 ? 2 : 1);`
`var barHeight = (byte) 100;`

`return new byte[] {29, 119, barWidth} // Width
                .AddBytes(new byte[] {29, 104, barHeight }) // Height
                .AddBytes(new byte[] {29, 102, 1}) // font hri character
                .AddBytes(new byte[] {29, 72, 0}) // If print code informed
                .AddBytes(new byte[] {29, 107, 73}) // printCode
                .AddBytes(new[] {(byte) (code.Length + 2)})
                .AddBytes(new byte[] {123, 66})
                .AddBytes(code, encoding);`

Um outro detalhe importante, alguns modelos de impressora mais antigos não imprimem 44 caracteres no código de barras, se sua impressora for muito antiga, recomendo a substituição da mesma.

ficodigital, 2021-11-30:
@Dufayel , antes de mais nada obrigado pelo retorno!

Implementei a sugestão, porém sem sucesso; seguindo o último detalhe que você deu indicou, tenho o seguinte cenário de equipamentos:

[image]

Ambas com 42 colunas.

Sabe se tem alguma outra alternativa ou só com a troca dos equipamentos?

Dufayel, 2021-11-30:
@ficodigital infelizmente a epson TM-T88V é um dos modelos que não consegui imprimir código de barras com 44 caracteres, quanto a sweda, não sei lhe informar. 
Aqui na empresa as impressoras que menos dão problema e geralmente recomendamos para nossos clientes são as epson TM-T20X.

ficodigital, 2021-11-30:
@Dufayel ,
Obrigado pela troca de ideias, vou agitar aqui pra ver quais serão os próximos passos...rsrsr abraço!

Dufayel, 2021-11-30:
Não sei se é para impressão de comprovantes fiscais que você pretende utilizar o código de barras, imagino que sim pela quantidade 44 caracteres, em alguns lugares já recebi comprovantes com o código de barras quebrado em duas colunas, 22 dígitos em cima + 22 dígitos em baixo.

Abaixo um exemplo.

[image]

leandrovip, 2021-12-01:
Olá, tudo certo?

Em todos os modelos listados no readme foram testados a função de impressão do código, mas é como o @Dufayel comentou, pode ser que necessite de alguma atualização do firmware da impressora para funcionar corretamente. De outra forma, creio que seria apenas testando mesmo.

No exemplo deixei com 22 caracteres porque é o padrão que utilizo aqui. Na função aceita mais, inclusive em alguns modelos imprime sem precisar de modificações os 44 caracteres da chave de acesso, mas como sei que nem todas aceitam (no padrão que está) eu deixo como 22 caracteres e faço a impressão com este split.

Se eu puder auxiliar em algo, só dizer.

[]´s

ficodigital, 2021-12-02:
> Não sei se é para impressão de comprovantes fiscais que você pretende utilizar o código de barras, imagino que sim pela quantidade, 44 caracteres, em alguns lugares já recebi comprovantes com o [...]

@Dufayel ,

Exatamente isso, será para impressão da chave CFe, olhando o manual de orientação do SAT também entendo que eu poderia gerar em duas linhas esse código de barras da chave CFe, cada um com 22 caracteres, mas é ai que mora o meu problema. 

Envio com 22 caracteres: printer.Code128("3519036109900800014100"); -> Não imprime o código de barras.

Envio com 20 caracteres: printer.Code128("35190361099008000141"); -> Imprime o código de barras.

O que eu não consegui entender ainda é porque enviando 22 caracteres não imprime o código de barras e enviando 20 caracteres ele imprime, como a impressora que tenho aqui é de 42 colunas não estou tentando gerar um único código de barras, desde o início estou na tentativa de gerar 2 códigos de barra conforme o exemplo perfeito que vc mandou.

ficodigital, 2021-12-02:
> Olá, tudo certo?
> 
> Em todos os modelos corrigidos no readme foram testados a função de impressão do código, mas é como o @Dufayel comentou, talvez por ser um modelo mais antigo seja necessário a atualização do firmware da impressora, se estiver disponível, para funcionar corretamente. De outra forma, creio que seria apenas testando mesmo.
> 
> No exemplo deixei com 22 caracteres porque é o padrão que utiliza aqui. Na função aceita mais, inclusive em alguns modelos imprime sem precisar de modificações de 44 caracteres da chave de acesso, mas como sei que nem todas as aceitam (não padrão que está) eu deixo como 22 caracteres e faço a impressão com este split.
> 
> Se eu puder auxiliar em algo, só dizer.
> 
> [] É

@leandrovip , tudo bem!?

Antes de mais nada obrigado pelo retorno!

Conforme o exemplo que o @Dufayel passou, estou tentando gerar o código de barras para a chave CFe (44 caracteres), como minha impressora é de 42 colunas, quebrei a string do código de barras em 2, ou seja, 22 caracteres para cada código de barras, o manual de orientação do SAT permite essa ação, porém:

Envio com 22 caracteres: printer.Code128 ("3519036109900800014100"); -> Não imprime o código de barras.

Envio com 20 caracteres: printer.Code128 ("35190361099008000141"); -> Imprime o código de barras.

Não sei como funciona essa dinâmica entre 42 e 44 colunas, porém entendo que pelo menos enviando 22 caracteres para gerar o código de barras, deveria funcionar; o que está martelando na minha cabeça é exatamente isso porque gera com 20 caracteres e não gera com 22 caracteres?

Uma solução paliativa que fiz foi a seguinte, gerar uma imagem do código de barras e depois chamar o método "vipPrinter.Image()".

[image]

Esse foi o resultado:

[image]

Mas ainda assim não acredito que seja a melhor opção, mas foi a única forma que consegui com a impressora que tenho aqui, mas ainda está martelando na minha cabeça porque gera com 20 caracteres e não gera com 22 caracteres no método vipPrinter.Code128()? Não acredito que seja um bug, mas gostaria de entender o motivo...

Dufayel, 2021-12-02:
Boa tarde @ficodigital , não tenho embasamento técnico, mas creio que seja limitação de hardware, sei que existe uma relação entre a largura das barras e a quantidade de caracteres que podem ser impressos.

Essa ideia de utilizar imagem é ótima, não tinha pensado em fazer dessa forma, se me permite vou utilizar dessa ideia para passar a imprimir em modelos que hoje não consigo, como essa TM-T88V do seu exemplo.

ficodigital, 2021-12-02:
> Boa tarde @ficodigital , não tenho embasamento técnico, mas creio que seja limitação de hardware, sei que existe uma relação entre a largura das barras e a quantidade de caracteres que podem ser impressos.
> 
> Uma possível solução é atualização do firmware, conforme sugerido pelo @leandrovip, pois por vezes na época de lançamento do equipamento não existia a demanda para tal tipo de impressão.
> 
> Essa ideia de utilizar imagem é ótima, não tinha pensado em fazer dessa forma, se me permite vou utilizar dessa ideia para passar a imprimir em modelos que hoje não consigo, como essa TM-T88V do seu exemplo.

@Dufayel ,

Imaginei algo desse tipo, mas tbm não tenho um conhecimento aprofundado a nível de hardware desses equipamentos.

Quanto utilizar a implementação via imagem, fique a vontade, na verdade fico feliz que vai resolver um cenário aí pra vc!

leandrovip, 2021-12-02:
@ficodigital, 

Chegou a baixar o software utilitário da impressora? [Aqui][(url](https://epson.com.br/Suporte/Ponto-de-venda/Impressoras-t%C3%A9rmicas/Epson-TM-T88V-Series/s/SPT_C31CA85011?review-filter=Windows+XP+32-bit)) ou [Por aqui](https://download.epson-biz.com/modules/pos/index.php?page=prod&pcat=3&pid=36)

- Na impressão com as configurações da impressora, qual a versão do firmware é impressa?
- Ao imprimir o código de barras diretamente pelo utilitário da epson, acontece o mesmo cenário? 

ps. Sobre o código como imagem, ficou bacana. Mais uma forma de resolver rs.

[]´s

closed 2021-12-12 by leandrovip