leandrovip/Vip.Printer #51: Impressão por IP

marciolevorato, 2022-01-05:
Olá. Seu projeto é bem prático e objetivo, parabéns.
Estou somente com dificuldade de imprimir quando a impressora está na rede, só foi possível imprimir quando a impressora está instalada no Windows da máquina e com o nome certo. 
Há uma forma mais prática de fazer isto?

brasvidal, 2022-01-05:
Você consegue fazer a impressão via impressão de teste da impressora compartilhada do windows?

teste de impressão mesmo.

marciolevorato, 2022-01-05:
Bom dia, teste de impressão via impressora compartilhada, sim.

Em outros projetos, não tão práticos quanto o VIP.Printer, é possível colocar uma linha de código com o IP e Porta da impressora que funciona.

brasvidal, 2022-01-05:
Então, já trouxe mais uma variável, agora você está falando de colocar IP e Porta, no post anterior era sobre impressora compartilhada.
Qual situação não está funcionando?

brasvidal, 2022-01-05:
using Vip.Printer;

/* 
Instancie a classe Printer

Para instanciar a classe é necessário dois parâmetros:
- printerName (**Nome da impressora, porta ou o caminho do compartilhamento**)
- printerType (Set de comandos que será utilizado. PrinterType.Bematech, PrinterType.Epson ou PrinterType.Daruma)
*/ 

var printer = new Printer("MP2500", PrinterType.Bematech); 

printer.WriteLine("Meu texto aqui!");
printer.PartialPaperCut();
printer.PrintDocument();

brasvidal, 2022-01-05:
var printer = new Printer("192.168.1.150:09100", PrinterType.Bematech);

marciolevorato, 2022-01-05:
var printer = new Printer("192.168.10.102:09100", PrinterType.Epson);
var printer = new Printer("192.168.10.102:9100", PrinterType.Epson);

é uma Elgin i9. 
Eu só consegui imprimir na impressora deste IP e porta após instalá-la na máquina e adicionar ali no campo o nome que ela está instalada na máquina.

[image]

[image]

leandrovip, 2022-01-06:
> Olá. Seu projeto é bem prático e objetivo, parabéns. Estou somente com dificuldade de imprimir quando a impressora está na rede, por IP e porta, só foi possível imprimir quando a impressora está instalada no Windows da máquina e com o nome certo, não colocando diretamente o IP e porta. Há uma forma mais prática de fazer isto?

Olá, @marciolevorato. Muito obrigado!

Atualmente não é possível realizar a impressão diretamente pelo IP e Porta. Porém, já estou implementando uma interface para esta finalidade, para aceitar tanto IP/Porta diretamente quanto outras opções de acesso se for o caso.

Uma alternativa paliativa que utilizo quando preciso acessar a impressora via rede é acessa-lá através do 'nome do compartilhamento' da impressora (ex: \\pc\impressora). Mas ainda sim é necessário realizar a instalação da mesma em um computador.

Um teste que você pode tentar executar, é informar no caminho da impressora algo como \\192.168.10.102 e ver se acessa corretamente. Se puder, me avise o resultado por favor.

Creio que logo consigo fazer o merge das atualizações com esse tipo de possibilidade, de forma mais direta. Na próxima semana possivelmente.

[]´s

marciolevorato, 2022-01-06:
> Olá, @marciolevorato. Muito obrigado!
>
> Atualmente não é possível realizar a impressão diretamente pelo IP e Porta. Porém, já estou implementando uma interface para esta finalidade, para aceitar [...]

Que legal que isto já está em desenvolvimento! Se tiver algo que possamos ajudar é só falar.
Bom, quanto a botar o caminho da impressora ao declará-la, fiz alguns testes, sem sucesso. Só funcionou após eu colocar o nome da impressora instalada, fazendo as etapas abaixo, no Windows 10:

Configuração de impressão deve estar no mesmo domínio. CMD > IPCONFIG
1- Instalar ELGIN Printer Driver_v-1.6.6, colocar Elgin i9, porta TCP/IP
1.1- Adicionar Porta TCP/IP, colocar o IP da impressora

2-Ir em Impressoras e Scanner
2.1-Adicionar uma impressora ou scanner
2.2-A impressora que eu quero não está na lista
2.3-Adicionar uma impressora local ou de rede usando configurações manuais
2.4-Usar uma porta existente + SELECIONAR A PORTA COM O IP

3-Editar o nome da impressora instalada e utilizá-lo no projeto.

leandrovip, 2022-01-06:
Olá, 

Geralmente não falo instalação dos drivers específicos da I9, instalo como Generic Only Text mesmo. Mas penso que apenas este processo diferente não alteraria o resultado.

Tenho uma impressora aqui comigo e vou fazer alguns testes sobre isso.

De qualquer forma, assim que eu subir as alterações linko aqui na issue para você poder fazer um teste, se possível.

[]´s

closed 2022-04-18 by leandrovip