leandrovip/Vip.Printer #55: Problemas com palavras com assento

meciasbueno, 2022-03-08:
Estou tendo problemas com palavras com assento, está imprimindo um caracter "especial" no lugar.
Como resolvo isso? no meu caso é português.

leandrovip, 2022-03-08:
Olá! 

Qual o modelo impressora? 
Qual o set de comando utilizado? 
Imprime o auto-test da impressora e veja qual é o charset configurado?

Obs: Geralmente é por conta do charset ou pelo encoding do conteúdo.

[]´s

meciasbueno, 2022-03-08:
> Olá!
> 
> Qual o modelo impressora? Qual o set de comando utilizado? Imprime o auto-test da impressora e veja qual é o charset configurado?
> 
> Obs: Geralmente é por conta do charset ou pelo encoding do conteúdo.
> 
> []´s

Está ocorrendo na minha impressora de teste (Portatil Mobile Printer), nesta eu não consegui imprimir o auto test, não sai nada. Na especificação dela aparece ANK: caracteres ANK.

Na impressora do cliente também está acontecendo problemas com assento, é a Bematech MP 4200 TH

leandrovip, 2022-03-09:
Na sua impressora de teste, acredito que seja por questão do chartset/encoding. O conteúdo que está enviando está como? UTF8? ASCII?

Na MP4200, você pode conferir a configuração imprimindo o auto-teste ou através do software 'Bematech User Software', pois se for preciso você consegue alterar. 

Geralmente deixo configurado nestas impressoras o charset IBM850 (salvo engano)

closed 2022-03-18 by leandrovip