leandrovip/Vip.Printer #61: Net core e Net 6

marcelomera, 2023-01-04:
Olá

Poderia verificar de atualizar seu pacote para os Net core e Net 6 tenho projetos em ambos. 

Obrigado.

leandrovip, 2023-01-04:
Boa noite,

O projeto tem compilação para .Net Standard 2.0

Houve algum problema para executar?

marcelomera, 2023-01-04:
Fiz o teste e deu certo, por acaso você tem um exemplo de como colocar o codigo do produto, nome do produto, qtde e valor tudo na mesma linha porem cada um alinhado na sua "coluna" ?

desde ja obrigado e parabens pelo projeto, irei fazer alguns testes mais tarde.

leandrovip, 2023-01-04:
Que bom que deu certo. 

> Fiz o teste e deu certo, por acaso você tem um exemplo de como colocar o codigo do produto, nome do produto, qtde e valor tudo na mesma linha porem cada um alinhado na sua "coluna" ?

Não tem no projeto de demonstração, porém depois vou subir um de cupom, por exemplo. Aí fica fácil fazer ajustes ou até mesmo alguma ideia sobre como fazer.

Obrigado pelo feedback, qualquer coisa só avisar 👍

closed 2023-01-05 by marcelomera