leandrovip/Vip.Printer #62: Encoding.

marcelomera, 2023-01-28:
Somente hoje fui conseguir testar o pacote.

Primeiro que não consegui localizar a impressora Elgin no PrinterType, usei todas as opções e em todas elas apresentou o seguinte erro:

[image]

System.NotSupportedException: 'No data is available for encoding 850. For information on defining a custom encoding, see the documentation for the Encoding.RegisterProvider method.'

Pode me orientar qual o procedimento correto.

leandrovip, 2023-01-28:
Qual a marca e modelo da impressora?
Elá está utilizando qual set de comando?

- Caso for da linha ELGIN (I8, I9..) você utilizará o _PrinterType = Epson_, que é a configuração para o set de comando ESCPOS.

- Caso estiver utilizando o projeto em .net core+ pode ser necessário que você coloque esta linha abaixo na inicialização do projeto, ou seja, nas classes program/main/startup etc..

`Encoding.RegisterProvider(System.Text.CodePagesEncodingProvider.Instance);`

marcelomera, 2023-01-30:
É uma elgin i9, 

Ao colocar o parâmetro na classe program do projeto parou de apresentar erro porem a impressora só reimprimi depois que desliga e liga.

Como faz para ver o set de comando que mencionou ?

leandrovip, 2023-01-31:
Para ver o set de comando:

- Desligue a impressora
- Segure pressionado o botão feed
- Ligue a impressora

Aguarde até começar a imprimir, veja no comprovante impresso as instruções para imprimir as configurações da impressora.

Está utilizando ela instalada via spooler do windows?
Usando o driver, Generic Text-Only ou o driver específico da Elgin?

Se precisar, me chame aqui no [Whatsapp ](https://api.whatsapp.com/send?phone=5517991765035) que lhe ajudo a testar remotamente.

[]´s

marcelomera, 2023-02-05:
Estou utilizando via Spooler e o driver da impressora.

Anexo impressão 
[image]

leandrovip, 2023-11-13:
Fechando a issue, caso ainda precise de ajuda, pode reabrir.

closed 2023-11-13 by leandrovip