leandrovip/Vip.Printer #63: Tamanho do código de barras Ean13

ptrick0, 2023-01-31:
Estou utilizando uma impressora Bematech MP-2500 TH (Esc/Pos), gostaria de saber se tem alguma forma de aumentar o código de barras, ele ocupa apenas metade do papel que estou utilizando (80mm).

leandrovip, 2023-02-01:
Consegue mandar uma foto?

Teria que dar uma olhada no manual de comandos para ver se tem essa possibilidade, eu imagino que não. De qualquer forma assim que eu conseguir posso fazer uma pesquisa.

Dufayel, 2023-02-01:
> 

Bom dia, o tamanho vertical do código de barras depende da quantidade de dígitos que ele representa, tente gerar códigos com diferentes tamanhos e verá a diferença. 

Há um tempo atrás tive bastante problemas com a impressão de códigos de barras, na época conversei com o pessoal aqui do projeto e me sugeriram enviar o código de barras como imagem para a impressora, assim poderia manipula-lo conforme minha necessidade, essa abordagem funcionou muito bem em meu projeto.

ptrick0, 2023-02-02:
Sem problemas, era mais uma curiosidade caso houvesse a necessidade de aumentar. O leitor que utilizo para leitura funcionou perfeitamente.
Caso no futuro eu precise vou tentar a abordagem que @Dufayel sugeriu! Muito obrigado aos dois!

closed 2023-02-02 by ptrick0