leandrovip/Vip.Printer #65: Dúvida sobre encoding (acentos, acentuação)

EwertonDutra, 2023-04-12:
Boa tarde, 
Como vai?
Estou com uma dúvida porque a impressão na Elgin I9 está saindo com caracteres estranhos (nas acentuações e ç)

Atualmente estou usando assim:
```
var printer = new Printer(dadosXML.GetNomeImpressora(), printerType, encoding: Encoding.GetEncoding(dadosXML.GetEncoding()));

public string GetEncoding()
        {
            string encoding = xml.SelectSingleNode("info").SelectSingleNode("encoding").FirstChild.InnerText.ToString();

            // Se vazio padrão é: ISO-8859-1
            if (string.IsNullOrEmpty(encoding))
            {
                encoding = "ISO-8859-1";
            }

            return encoding;
        }
```

Nesse caso acima estou sempre usando o "ISO-8859-1"
**Seria esse mesmo a usar ou tem outro para acentuação sair correto?**

_Outras impressões sai correto na impressora, só do meu programa que sai assim_

leandrovip, 2023-04-12:
Boa tarde,

Geralmente na I9 você não precisa de passar encoding, a não ser que seja para algo específico como: encoding da impressora diferente do padrão, encoding dos dados enviados fora do UTF8, algo do tipo.

Tente deixar o encoding padrão da propria biblioteca para fazer um teste.

[]´s

leandrovip, 2023-11-13:
Fechando a issue, caso ainda precise de ajuda, pode reabrir.