leandrovip/Vip.Printer #66: Daruma Image

reislanes, 2023-08-24:
Ao usar o Tipo de Impressora Daruma, a impressão da imagem fica apenas alinhado à esquerda, não consigo centralizar ou alinhar a direita.

leandrovip, 2023-08-24:
OIá, como vai?

Qual o modelo da impressora?
Consegue gerar uma impressão de teste através do método: **TestPrinter()** ? Neste teste a impressão sai alinhada corretamente?

Pode ser que:
- os comandos de alinhamento para este modelo de impressora em específico não estejam de acordo;
- os comandos de alinhamento não estejam corretos para o set de comando da daruma;

É possível configurar o set de comando da impressora para EscPos para fazer um teste?

Como não tenho nenhuma impressora deste modelo, fico um pouco limitado em ajudar, mas com estas questões acima conseguimos ter um norte (pelo menos).

[]´s

reislanes, 2023-08-30:
O modelo é DR700. Eu precisei deixar no cliente, mas vou fazer os testes e postar aqui.

leandrovip, 2023-08-30:
Oks, fico no aguardo

[]´s

leandrovip, 2023-11-13:
Fechando a issue, caso ainda precise de ajuda, pode reabrir.

closed 2023-11-13 by leandrovip