leandrovip/Vip.Printer #69: Problem with Cyrilic codepage when in Condensed mode

pajicdejan, 2024-10-29:
Test printer Birch CP-Q3, code page 1251-Cyrilic(Windows) - printing in Normal mode is totally OK but any attempt to print Cyrilic characters in condensed mode, results with blank spaces or question marks instead of wanted words. Following code:
Vip.Printer.Printer prn = new Vip.Printer.Printer("CP-Q3", Vip.Printer.Enums.PrinterType.Epson, Encoding.GetEncoding("Windows-1251"));
prn.CondensedMode(Vip.Printer.Enums.PrinterModeState.On);
prn.WriteLine("=====ОВО ЈЕ ЋИРИЛИЦА====="); // This means "THIS IS CYRILIC"
prn.CondensedMode(Vip.Printer.Enums.PrinterModeState.Off);
is resulting with: "===== =====" // Empty space in the middle is actually 15 blanks wide (length of my phrase) - GitHub shortens it when message is saved.
Than tried CondensedMode with string constructor, as well as Font method - both ways give question marks for each character wanted to print. Like:
prn.CondensedMode("=====ОВО ЈЕ ЋИРИЛИЦА=====");
prints: "=====???????????????====="); // 15 question marks
Tested it for few hours and concluded that problem emerges with characters/code pages other than IBM Standard, like 1251-Cyrilic or 1250-Central Europian (characters like ŠšĐđŽžČčĆć), both lower or upper case.
Any idea how to overcome this issue? Is it because of driver or some strange hardware fault?

leandrovip, 2025-04-05:
Hello, sorry for the delay. Are you still having this problem?

The default encoding is IBM, but the library allows you to change this when the class is instantiated.

Anyway, I think the first step would be to check the encoding configured on the printer. It is usually not necessary to use the printer's specific driver, unless this is essential for the printer to work.

Let me know if you still need help

leandrovip, 2025-11-04:
If you still need help, let me know and reopen this issue