leandrovip/Vip.Printer #70: Imprimindo número 5 no início da linha

diegodariolli, 2025-03-19:
Olá,
Estou implementando a biblioteca em nosso software porém está tendo um problema onde está imprimindo o número 5 no início de cada linha.
Estou testando em um Bematech MP4200 TH.

leandrovip, 2025-03-20:
Olá, qual set de comando está utilizando? A impressora esta com qual versão e qual set de impressão? Se possível anexa uma impressão com as configurações dela.

Na aplicação de demonstração ocorre o mesmo comportamento?

[]´s

diegodariolli, 2025-03-20:
Consegui resolver o problema. Em alguns lugares eu estava "ligando" o modo expandido juntamente com o condensado por isso apresentava esse problema.
Agora funcionou quase tudo, mas agora estou com problema na codificação em alguns caracteres como: Ô, Á, Ã.
Me parece que são os caracteres maiúsculos porque os minúsculos parecem estar funcionando todos.

diegodariolli, 2025-03-21:
Consegui achar o problema.
Eu estava usando condensado e expandido em alguns lugares por isso estava ocorrendo aquilo.
Agora estou com problema na codificação, algumas acentuações estão ficando incorretas.

leandrovip, 2025-04-05:
Ainda está com este comportamento?

diegodariolli, 2025-04-07:
Sim, ainda está.